SukkoPera/LittleSixteen #1: Looking over your schematics

bwack, 2020-02-02:
Hi SukkoPera ! I'm looking over your schematics, mainly just because of curiosity and interest :)

I found a mistake at page 11: C11 is connected on wrong side of L1.

I will notify you if I find more.

Hans/Bwack

bwack, 2020-02-02:
Page 10 (PLA): Addr_clk and KERN mixed up on U16 PLA.

bwack, 2020-02-02:
- Page 9 (power):
  - Looks like wrong power switch, and resting at ON position.
  - comment to R10 big white resistor. It provides about 200mA extra current. (9-5)V/20 ~ 200mA. It is very odd!
  - U11E and U9G on wrong page.
- Page 8: Exp conn:
  - BRESET ?
- Page 7 (cassette and disc drive):
  - The RP1 and CST WRT is shared with the disk drive signal on p6. Not clear in the schematics that p6 is shared, though I do see it now maybe because I have the original next to the new one.
  - p0..p7 on page 2 and 7: I would call them by signal name rather than cpu port io name in my opinion. I got confused.
- Page 6: I'll skip this page. It should be redrawn.
- Page 5 (ROM): OK
- Page 4 (RAM): OK
- Page 3 (TED and clock): OK
- Page 2 (CPU): OK

other:
Check pinout of all bipolars.
page 6 (keyb): I don't like this page. I like the original better. At least the P0..P7 part. I do like the comments though.
page 5 (ram): double check RP3 and RP4 connections. Original schematic is not clear here.
Page 10 (PLA): F0 and F3 looks like test/connection points in original schematics but not in new schematics. What is most correct?

Thats it. Overall, very nicely done!

SukkoPera, 2020-02-02:
Hi @bwack, it's a honor to have someone like you check my work :).

I will go through all your observations as soon as I am finished with the project I'm currently working on.

Thanks a lot, I really appreciate it!

SukkoPera, 2020-10-04:
I finally had a look on these, the fixes are in the fix_errors branch. In detail:

> page 11: C11 is connected on wrong side of L1.

Fixed.

> Page 10 (PLA): Addr_clk and KERN mixed up on U16 PLA.

Fixed.

> Page 9 (power):
>   Looks like wrong power switch, and resting at ON position.

This can be addressed in the PCB footprint (if ever).

>   U11E and U9G on wrong page.

Well, U9 and U11 are used in more than one page, so I felt the best place to put their decoupling caps was the Power page.

> Page 8: Exp conn:    BRESET ?

Since it is unnamed on the C16 schematics, I probably took this name from the +4 schematics, where it is defined as *Buffered System Reset*.

> Page 7 (cassette and disc drive):
>    The RP1 and CST WRT is shared with the disk drive signal on p6. Not clear in the schematics that p6 is shared, though I do see it now maybe because I have the [...]

As far as my KiCad knowledge goes, this is the best I can do. That's why I also added text labels clarifying the signal names.

> Page 6: I'll skip this page. It should be redrawn.
> I don't like this page. I like the original better. At least the P0..P7 part. I do like the comments though.

You are probably right, but I really don't feel like doing this at the moment. I'm all open for contributions though ;).

> page 5 (ram): double check RP3 and RP4 connections. Original schematic is not clear here.

This was traced by myself on a real board. I know it looks very odd but it simplifies the routing, that's why they did it like that, I guess.

> Page 10 (PLA): F0 and F3 looks like test/connection points in original schematics but not in new schematics. What is most correct?

I have no idea what *SCS* means for F0, while F3 is uses on the +4 as the ACIA CS. I have added in a note for the latter.

I think we can close this and open a new ticket for page 6, asking for help. Any comments?

SukkoPera, 2020-10-25:
I'm closing this as I fixed all I could. Feel free to open a new issue if you have any more comments :).

closed 2020-10-25 by SukkoPera

bwack, 2020-10-25:
Hi. Glad I could help :) I haven't had time to look at this again. I looked at page 6, and yes it is a lot of work to change it. What I didn't favor was the crossing. What commodore did there was to change the number order on the keyboard pin symbol.
And easy fix could be to remove all the wiring from the key connector and U13, use buses. I can try that if you like.

SukkoPera, 2020-10-25:
Sure, go ahead if you feel like it :).

bwack, 2020-12-22:
Hi Long time. I just remembered something about the work that I did on the schematics. Please use a larger grid, or try not to change the grid in schematics. A very simple job turned into a much harder one. Just saying.

SukkoPera, 2020-12-23:
I don't think I ever changed the grid size. Not willingly, at least.

I will (hopefully) have a look at your contribution during the holidays, sorry for the delay.

bwack, 2021-05-01:
Sorry about the misguided blame. I spoke with a guy on facebook. He said that in KiCad 4 there was a bug where the symbols had an 0.1mm offset to the grid, and it was fixed at a later point (causing more trouble hehe). At the time when I imported this project into kicad 5, it led me to believe you had used a tiny grid.