SukkoPera/LittleSixteen #8: using the 6502 instead

giobbino, 2023-01-02:
Hello!
Amazing project!
I was wondering about the chance to use a 6502 as CPU using this project:
https://github.com/monotech/MOS_CPU_Replacer

of course one could build the adapter, but maybe it would be fine to add it directly on the board (of course if there's enough space left on the board to do that...)

Well, just a thought...

SukkoPera, 2023-01-02:
That would be a very good idea, but:
1. It hadn't been released yet when I did this (or at least I was not yet aware of it);
2. It seems to still have a few compatibility issues;
3. It requires way more components than can fit on the current board.

I will definitely take it into consideration if the remaining issues are fixed AND if I ever switch this to SMD components, which would probably allow us to reclaim quite some space.

Thanks for your thoughts anyway 😉.

giobbino, 2023-01-02:
I understand. Also, I bought some boards from JLCPCB with the idea to have them ready on hand, just in case; however, since luckily everything still work fine here at home, I didn't test it yet - so I don't know if those adapters work fine or not. 
cheers!
P.S. sukkopera = succo di frutta alla pera? I love it too ;-)

SukkoPera, 2023-01-03:
Well, the author of that adapter seems to know what he's doing :). I still haven't tried it either but I am keeping an eye on it.

PS: That's it, but I actually hate succo di frutta alla pera :).