SukkoPera/LittleSixteen #11: BOM for Little Sixteen V3 Board

Snocksman, 2023-03-24:
Hi !

Is there a BOM for the little Sixteen V3 Board available ? I ordered some Boards but could not find a BOM yet.

Greetings
Mathias

SukkoPera, 2023-03-24:
Nope, haven't had time to put one together yet. Maybe you can help with it? You can start with what you can find in Kicad.

Snocksman, 2023-03-24:
OK, i´ll have a look at Kicad. So I have something to do at the weekend... ;D
When I got the BOM ready, you can add it to this cool project.

SukkoPera, 2023-03-24:
Yup, let's leave this open then!

Snocksman, 2023-03-26:
Hi !

I have found most of the Components, but for some i need a few more informations/values (maybe an example part):

CT1
D12
EM3, EM4, EM5
FB2, FB3, FB4,...
FB12, FB13, FB14,...
L1
L81

Thank you !
Best,
Mathias

SukkoPera, 2023-03-28:
In our test build we used DSS1NB32A101Q91A for all EM* and FB*, totalling 51 of them. These filters are better than ferrites, but they also used to be more expensive, so Commodore only used them where essential, falling back to ferrites wherever possible.

These days I can't seem to find any reasonable ferrite that is significantly cheaper, so using filters everywhere is probably the best option. This one MIGHT do, but you're not going to save much anyway:

https://www.mouser.it/ProductDetail/Fastron/BEAD-4-900R-02

@kinmami: Any thoughts about that?

RP1, RP2 and RP5 are BUSSED.
RP3 and RP4, RP90 and RP91 are ISOLATED.

Other P/Ns from our test build:
CT1: EW Electronics GKG50015
D12: NZX6V8D,133 (Make sure to see the Errata page on the Wiki)
L1: Bourns 9250A-122-RC
L81: TE Connectivity SC30100KT

kinmami, 2023-03-28:
https://www.digikey.com/en/products/detail/laird-signal-integrity-products/28L0138-80R-10/1015875

https://www.mouser.it/ProductDetail/Laird-Performance-Materials/28L0138-80R-10?qs=qgZcrvoyHAd98f%252B0tLOU0Q%3D%3D

SukkoPera, 2023-03-28:
I'll leave the word to @kinmami :).

kinmami, 2023-03-28:
> Thank you for your answers ! Three questions are left:
> 
> I found these here at rs-online.com: DSS1NB32A271Q91A This should be a better equivalent for the Filters found on the original Commodore Boards...? => https://de.rs-online.com/web/p/signalfilter/2166093
> 
> Do you think this one will work for L81 ? https://www.reichelt.de/festinduktivitaet-axial-micc-ferrit-10--l-micc-10--p86427.html?&trstct=pol_1&nbc=1
> 
> I forgot to ask for CR100... Will this one fit ? https://www.reichelt.de/brueckengleichrichter-400-v-1-5-a-b280c1500rund-p181715.html?&nbc=1

Yes I think all of those above are compatible, thanks

msuontausta, 2023-03-30:
Here's our list: https://docs.google.com/spreadsheets/d/1dDgSGAWIuUnK9-j-XhQTBRzt-erFqXx8YhXyfMe616A/edit?usp=sharing

It's a bit rough around the edges and comments are in Finnish but all of the necessary information should be there. :)

SukkoPera, 2023-03-30:
Yeah! Thanks a lot! 👍

Snocksman, 2023-04-01:
@msuontausta Hey, thats cool. I could put my Reichelt order Information next to it, but I have only read permissions on the sheet.

Another question (maybe not really right to that issue): I got a W27C512. Is there any Information in which order i have to put the Basic,Kernal,Char-ROMs to the EEPROM ?

SukkoPera, 2023-04-01:
Yes, if you plan to use the all-in-one ROM, have a look at [this](https://github.com/SukkoPera/LittleSixteen/wiki/V3-Assembly-Notes#roms).

Actually make sure to read the whole document, and take care of the erratas.

SukkoPera, 2023-12-05:
You need to salvage it from an old board unfortunately :(. In V5 we'll turn it to the same connector as the C64, at least that one is still around.

kinmami, 2023-12-31:
Hello.
I have personally tested the listed parts without any problems, so I'm not certain about the root cause.
However, there is a different part number available with a capacitance of 270pF instead of 10000pF. You can try using this alternative part and let us know if it resolves the issue. Here is the link to the product: Murata Electronics DSS1NB32A271Q91A [](https://www.mouser.co.uk/ProductDetail/Murata-Electronics/DSS1NB32A271Q91A?qs=sGAEpiMZZMsFjaecyxH1mx6l0J9iAOtyVRV%252BvTeltWc416%2FtV58Y7w%3D%3D)

kinmami, 2023-12-31:
I have tested PAL machines with different custom chips.
Please try with this ferrite https://www.digikey.com/en/products/detail/kemet/B-01-RS/10442402
It is designed for signal / audio line filtering.

kinmami, 2024-01-17:
> Hi again, I just wanted to report back on my progress, in case it is helpful to folks. I have now removed all of the original 0.01uF DSS1N filter caps listed in the BOM, and replaced them with the following...
> 
> EM3, EM4, EM5 - DSS1NB32A271Q91A 270pF filter cap: https://www.mouser.co.uk/ProductDetail/Murata-Electronics/DSS1NB32A271Q91A?qs=0Ys4hG7ORMdMh3s4a3P7nA%3D%3D (Same capacitance value as is on original C16 boards)
> 
> FB1...FB95 - 28L0138-80R-10 86 Ohm @ 100MHz ferrite bead: https://www.mouser.co.uk/ProductDetail/Laird-Performance-Materials/28L0138-80R-10?qs=sGAEpiMZZMukHu%252BjC5l7YW%2FVzMlcaO9tAGwosoIUTdg%3D (they were about a quarter of the price), and again they seemed close enough to what was on the original C16 boards)
> 
> I know this means that the boards I built will have less filtering than the ones you built to test your designs but all of the problems I was experiencing have now gone. Both keyboards I have now work properly on both new boards, both the original and the replacement the PLAs seem to work properly and the machines start up every time, the games and demos with fastloaders now load and work from my PI-1541, and the intermittent wobbly video picture I had, which was also reported by others, has gone too.
> 
> The board design itself works great, it seems (for me at least) it was just the filter caps causing all of the issues I was experiencing.
> 
> Next stop for me is to build some of your expansion cartridges and test those out too.
> 
> Thanks very much for all of your hard work and your help, I am a very happy amateur hardware builder! :)

A word of appreciation for your troubleshooting and reporting. We are going to update the BOM list accordingly.

SukkoPera, 2025-01-02:
Information integrated in the [V3 Assembly Notes](https://github.com/SukkoPera/LittleSixteen/wiki/V3-Assembly-Notes).

(At last!)

closed 2025-01-02 by SukkoPera

Jartza, 2025-01-27:
Hi!

I finally got time to assemble my LittleSixteen and indeed there was an issue with BOM me and @msuontausta  provided. Accidentally the EMI-filter was 10000pF instead of 100pF... these will definitely NOT work, if you have gotten the 10000pF (10nF) filters, do not assemble them but use ferrites instead!

I have updated the BOM sheet and removed link to wrong EMI filter, instead linking to 270pF version (as 100pF doesn't seem to be available in Digikey), and added link to ferrite beads.

Sorry about the confusion.