SukkoPera/LittleSixteen #12: Board is ready, but all I get is a black screen...

Snocksman, 2023-04-13:
Hi !

I got my Little Sixteen Board ready today and powered it up for the first time but all I got is a black screen. So I read the Wiki again and found out that I had to remove D93 and R93 if I want to use an original 8501 CPU... So I cut off both and tried again but still black screen... So I tested the CPU (maybe I killed it, because of the installed D93 and R93...), but it still works fine in my Plus4.

Is there something that I should do with the jumper pads JP1 & JP2 if I install 64KB of Memory ??? I found nothing about them in the wiki...

Snocksman, 2023-04-13:
OK, I got the Board running. But I have a Problem with the Colors and Picture Stability on my Upscaler.

Here is the picture how it looks like on my original C16:
[image]

and here how it looks like on my Little Sixteen:
[image]

About picture stability: The picture stays stable on the upscaler for about a minute, after that it begins wobling in one direction... looks like the characters in the picture start "walking" to the left or to the right...

Without the upscaler (connected to an old CRT), the problem with picture stability is gone, but the color problem is still there.

Any Idea about that ?

SukkoPera, 2023-04-14:
If there's anyone that can help you, that's @kinmami :).

Snocksman, 2023-04-14:
For further Information (Oszilloscope Pictures and so in) you can see my thread  on forum64: [Link](https://www.forum64.de/index.php?thread/135873-little-sixteen-board-c16-plus4-starthilfe/&postID=1984549#post1984549)

kinmami, 2023-04-14:
I assume the video cable is 75 ohms and the termination at the CRT monitor (or video upscaler+LCD) is also 75 ohms.
Pls verify on your side if the cable has a series resistor anywhere from 300 to 2K ohm on CHROMA (some cables originally made for the C64 have a series resistor). 
Said that, from your scope screen captures I can see the LUMA signal is too intense and it is saturating.
Would you please adjust the trimmer or changing the resistor at the video circuit section to achieve something close to 1Vpp (LUMA) and 1Vpp (CHROMA)?
Hope it helps.

SukkoPera, 2023-04-15:
If calibration of the pots is an issue, note that the wiki has instructions on how to to use fixed resistors in their place, still providing the best output quality, thanks to relentless testing by @kinmami :).

Snocksman, 2023-04-15:
In the cable I use there is a 325 Ohm Resistor in the Chroma connection.
I was able to get a good picture using the trimmers, but the picture is getting "unstable" after a minute, when i´m using the Upscaler. When I turn the trimmer RV2 counterclockwise as far as possible (I get a very dark picture) the picture gets suddenly stable again.

SukkoPera, 2025-01-04:
Video circuit updated in V4

closed 2025-01-04 by SukkoPera