SukkoPera/LittleSixteen #13: Working with single 27C246

anarkiwi, 2023-07-16:
Hello,

Thank you so much for this wonderful project. I have just built a V3 board and got everything working - great!

I was able to make my V3 board work with just a single 27C256 (because I had trouble sourcing 27C512s) - perhaps a minor modification could be made to make this easier?

After some experimentation I found that I could simply burn kernal + basic to a 27C256, configure the V3 board for "single ROM", and install a single 27C256 instead with the Vpp pin lifted/grounded. It looks then like the C16 kernal comes up, does not find the function ROMs and just drops into regular C16 mode. I have no plans ever to use the function ROMs so this is fine for me.

Maybe an additional jumper could be added to allow the ROMs A15/Vpp to be grounded? 

Thanks again for your fantastic work,

SukkoPera, 2023-07-16:
Good suggestion, even if AliExpress is literally full of W27C512 ;). I will have a look and probably do something in this regard anyway. I just don't understand why you suggest grounding the VPP pin, I think it should be tied to VCC for read-only operation. I'll check and let you know ASAP.

Thanks for taking your time to suggest an idea!

SukkoPera, 2023-07-16:
I've checked and I am now sure that VPP must be tied to VCC for normal read-only operation.

This is luckily very easy to do: just put pin 1 back into the socket and solder JP6 in the _Separate ROMs_ position (i.e.: left and center pads). Leave JP5 and JP7 in the _All-In-One_ position. This should work, let me know!

[image]