SukkoPera/LittleSixteen #15: V4

kinmami, 2024-02-23:
U91~95 lengthen the pad footprint and add a DOT on pin 1

kinmami, 2024-02-23:
CT1 enlarge the pad holes

kinmami, 2024-02-24:
CN6 pinout is wrong

kinmami, 2024-02-25:
to replace power resistors R91, R92, R94, R95 and R39, R40 with polyfuses (BOM list updated with the commercial p/n)

kinmami, 2024-02-25:
R32 for MOS 6581: to align the schematic and the interactive BOM list

gordonfpanam, 2024-09-10:
> R32 for MOS 6581: to align the schematic and the interactive BOM list

MOS 6581 as in a SID chip? Is this an attempt to integrate the SID for Psytronix TED titles that support SID expansion carts?

SukkoPera, 2024-09-10:
Yes, V4 has a built-in SIDcard. See https://plus4world.powweb.com/forum/47104.

closed 2025-01-04 by SukkoPera

gordonfpanam, 2025-01-05:
I'll be looking for the PCBWay link to order v4 boards as soon as you have it up.

SukkoPera, 2025-01-05:
@gordonfpanam: That's coming in a few days, but if you want I can send you the gerbers and you can order the board on JLC, which should be cheaper.