SukkoPera/LittleSixteen #17: No NTSC Colour on v4

gordonfpanam, 2025-03-30:
I'm not getting any colour on NTSC with a 14.31818 crystal and NTSC kernal. Depending on the monitor, I could get either no colour at all in NTSC or PAL, colour in PAL but no colour in NTSC, or colour in PAL but flickering between colour and monochrome in NTSC.

I've already worked back through the video amp circuits, effectively converting both circuits into v3 form, but keeping R86 as 1k to avoid drawing too much current from the TED Sync/Luma pin. I want to work forward toward v4 again until I lose PAL colour, but I've run out of time for now.

I've gathered scope captures of the luma and chroma signals and I'll try to put together a side-by-side comparison video. I've done this for the L16 v4 and for an original Commodore 16 to compare. The original C16 has very hot chroma signals but these work on any monitor in either PAL or NTSC. By comparison, when I do get colour PAL video using the Rev 3 circuits I can dial things back to 1V peak to peak (1VPP) on luma and 0.3VPP on chroma, which is supposed to be s-video spec according to The Retro Channel:

https://www.youtube.com/watch?v=agDFLPP9yIw

Doing the 1VPP luma and 0.4VPP chroma also removed checkerboarding and jailbars on the v4 board's LCA / s-video output, at least on a PAL-capable monitor that displayed colour.

To reproduce the problem you'll need a NTSC-capable monitor, crystal and kernal, and the current v4 video amp circuits as of RC16. JP18 didn't need to be set at all, in fact leaving it unset will clean up the s-video output some. This seemed to happen regardless of whether I installed RV1 and 2 or not, though having RV1 and 2 helped me tune the output.

SukkoPera, 2025-03-30:
There is no risk in changing - or even temporarily removing - R86. It does not control how much current is drawn from the TED outputs.

I don't know if anyone has ever built an NTSC version of V3. It's definitely uncommon to have an NTSC-capable device here in Europe, so it's difficult to help with that.

In any case, we must tackle this issue rationally if we want to get anywhere. For a start, I suggest that you go 100% back to the V3 circuit, which has been proven working by several independent builders, so please switch to the following values and report back:

R81* = 3.3k
R82* = 1k
R83* = 5.6k
R84 = 150
R85 = 220
R86 = 470
R87* = 150
R88* = 150
R89 = 150
R90* = 150

C81* = 330p
C82 = 68p
C84 = NOT MOUNTED
C85 = NOT MOUNTED (but won't hurt anyway)
C86 = NOT MOUNTED
C87 = NOT MOUNTED

JP81 and JP82 CLOSED
JP18: 1-2

An asterisk indicates values that haven't changed between V3 and V4, so it should be sufficient to check that those parts have the correct value.

gordonfpanam, 2025-03-30:
Just some positive feedback for this whole process… this is my first scratch build for a modern retro PC and your support so far is excellent.

gordonfpanam, 2025-03-30:
How much trouble would it be to ship you a flat panel monitor capable of both NTSC and PAL with composite and s-video inputs? I could find something that just has a 12VDC or 5VDC power input that would let you use a locally sourced power supply.

I suspect such monitors are also readily available on eBay. It's where I got a few of mine. I sourced others locally from thrift stores and flea markets.

SukkoPera, 2025-03-31:
Thanks for the offer but no, at the moment I'd prefer to avoid that. You're going to be our reference for NTSC! :)

krzycho751, 2025-04-01:
Hi guys,
I recently built a V3 with an NTSC crystal and ROM. I can confirm it works!
I went with a simplified setup, leaving out RV1 and RV2.
The video signal isn’t great (visually — I haven’t hooked it up to a scope yet), but I suspect it’s due to a noisy environment and a less-than-ideal PSU. I’m getting wide bands of noise slowly drifting from bottom to top on the screen.
For display, I’m using a ~10 year old Samsung TV, which might also be contributing to the issue.

gordonfpanam, 2025-04-01:
Well this is embarrassing… 

I put in the v3 circuits as you asked but still got only monochrome. But on a lark I tried adjusting the variable capacitor at CT1 and I got colour NTSC.  

*facepalm*

It’s a bit dark compared to my prior tests, but I’ll put back the v4 circuits and try again. 

I think RP1 and RP2 are still important at least on this test build, but I’ll try to measure the effective resistances and see if they match your recommended values for omitting those.

So it appears a precise crystal clock is really important for colour video. My scope doesn’t measure high frequencies very well, being a cheap Hantek USB model, but maybe I can get a precise horizontal sync value at least.

gordonfpanam, 2025-04-02:
OK I determined that the original V4 circuit can do colour NTSC. It doesn't do it _very well,_ but it can do it.

The hybrid v3/v4 circuits I had two days ago seemed to do a much better job, but at least the current v4 circuit works well enough.

One thing I noticed that when I put the composite mixer back into service, JP18 towards the cartridge port worked much better than JP18 towards the user port. Both are pretty dirty, but JP18 towards the user port looked a lot like Apple II artifact colour when I had the monoscope generator program on screen 1 for that. I had fewer video artifacts, less checkerboarding, and less jailbars with JP18 set to the right and 

So for general purpose use, I'd recommend building out the v4 circuit, but with JP18 set to the right (towards cartridge port) but with RV1 and 2 installed, so a user can adjust the video signal amplitudes. When JP18 is set, Luma almost doubles in amplitude up to 2VPP and needs to be dialed back using RV2. Even without a scope, one can eyeball the picture quality and tune it using first adjusting CV1 to get colour, then RV1 for brightness, then RV2 for less checkerboarding and jailbars.

For a perfectionist that has access to an LCA or s-video monitor, and as long as they will use s-video exclusively, I would recommend a hybrid of the v3 and v4 circuits with the composite mixer and associated transistor completely omitted. With the composite mixer omitted you'd need to dial RV1 and 2 back up but you'd get a very clean picture. I'll write something up as a supplement (optional for perfectionists only) for the Wiki when I've come back to that.

SukkoPera, 2025-04-08:
I've added basic info about this to the wiki. Feel free to reopen to submit the info for perfectionists, I will move it to the wiki in turn.

closed 2025-04-08 by SukkoPera

SukkoPera, 2025-04-08:
If you want to add info for perfectionist, please start from this text I've just added to the wiki:

https://github.com/SukkoPera/LittleSixteen/wiki/Fine%E2%80%90Tuning-the-Video-Signal