SukkoPera/LittleSixteen #18: Potential Short Circuit with Metal Keyboard Frame

gordonfpanam, 2025-03-30:
On a 64c case with original metallic keyboard frames, there is a significant risk of short-circuiting unused surface mount pads for U93, U94 and U95. Metallic frames are outright unusable if these are installed instead of the through-hole diodes for electrostatic discharge protection.

Under recommended conditions, U93 through 95 would be installed instead of the diodes.

In addition, U17 and U18 sit too high on the board if socketed and an original metal frame is used. RP2 also doesn't allow the metal frame to sit correctly.

[image]

To workaround this, install U17 and U18 directly to the board, and modify the metallic frame. Use electrical tape around the mount point to prevent shorting against the pads for U93-95. Alternately, a 3D-printed plastic frame for a 64c keyboard may avoid this entirely.

[image]

SukkoPera, 2025-03-31:
Thanks for this report, I will move it to the Wiki ASAP.

SukkoPera, 2025-04-08:
Done.

closed 2025-04-08 by SukkoPera

gordonfpanam, 2025-05-05:
I don't need to re-open this specific issue, but at some point I'll try fitting a 64c short board RF shield on one of my two builds to see if there are any more potential short circuits or fitting problems. Today's US FCC isn't as paranoid about RF emissions, but I'm curious.