SukkoPera/LittleSixteen #19: Datasette Functionality on v4 without User Port?

gordonfpanam, 2025-04-02:
The v4 Assembly Notes include a section describing that the [Datasette port won't work](https://learn.microsoft.com/en-us/entra/identity/hybrid/cloud-sync/migrate-azure-ad-connect-to-cloud-sync) on the initial assembly. The v4 Configuration describes that the 6529 / 74x654 from the user port [controls some of the Datasette functionality](https://learn.microsoft.com/en-us/entra/identity/hybrid/cloud-sync/migrate-azure-ad-connect-to-cloud-sync) and references JP12.

Diag264 reports the Datasette port as bad, which matches the Assembly Notes and hasn't been a show-stopper yet. At what point should I expect this port to be working? When the user port section is complete? 

(If I get time I will try comparing this against the v3 Datasette port to see what the differences are.) #

gordonfpanam, 2025-04-02:
I think I found it, in that the v4 needs to have the user port section completed at least, as this is the plus/4 version of the datassette. My 74x654 parts are still in transit so I won't be able to validate this for some time.

SukkoPera, 2025-04-02:
You need U22 and JP12 in the UP position.

closed 2025-04-02 by SukkoPera