SukkoPera/LittleSixteen #20: JPs 81 and 82 shorted under solder mask

gordonfpanam, 2025-04-13:
I found it difficult to adjust luma and chroma levels on my second build, but discovered that the RV bypass jumpers for omitting RV1 and RV2 are shorted from the factory.

Inspecting the PCB layer used by these in KiCad confirms a possibly unintended short under the solder mask.

I had to cut the traces under the mask to use RV1 and RV2, which allow better control of luma / chroma / s-video output and reduces video artifacts. Maybe changing R87 and R84 to different values will get better results, but I’m pretty sure leaving these shorted wasn’t intentional.

SukkoPera, 2025-04-13:
This was actually by design, since we thought the easy way would be sufficient for most users.

But maybe it would be better to let the user close the jumper, I will think about it. In any case the instructions should be made clearer.

Oh, and I'm surprised that you didn't do this on your first build!

gordonfpanam, 2025-04-13:
I think I did the cut by accident the first time. I went to remove the solder from the JP8x pads to find the wick “wasn’t working,” then took a dremel to them.