SukkoPera/LittleSixteen #21: Question: 6581 or 8580 preferred for SID chip?

gordonfpanam, 2025-04-16:
I tried three 6581s across two L16 v4 builds and got different results for all three SID chips.

To compare the two I ran the Giana Sisters Demo for the plus/4 on the L16 and compared it to the real Giana Sisters game on the C64, and I also ran a demo whose name might trip some content filters here but uses digital audio playback.

The first SID sounded like trash, but sounded fine on a C64 longboard. The digital audio especially had some artifacts.
The second SID sounded almost perfect, in fact it's the same one in the YT Short you linked to in the Wiki.
The third SID was much better than the first, but still sounded off in the Giana Sisters Demo. Digital audio was much better. The initial loading screen ping for Lykia was also somewhat off on the third SID but much better on the second.

I was careful to make sure the SID area of the board was built out identical on both, using the correct voltage regulator for instance, for 6581s. I could swap in 8580s and the other regulator but I'd also have to swap in a few other discrete parts.

Considering the original SID card seemed to favour the 8580 if plus4world is to be believed, should I have gone with 8580s in these builds? Could there be that much of a difference between 6581s that they would work fine on a C64 longboard but have different behaviour on this build? Would this be more consistent using 8580s than 6581s?

As a side note, I know someone close by that has a 6581 that is iffy on a Commodore 128 flat system, but is fine on his C64 longboard as well. These are both supposed to use 6581s apparently - I checked the VSID pin on the 128 and it was +12VDC like it is for here.

SukkoPera, 2025-04-17:
I have never tested the Giana demo, but it has to be expected that SID sound on C16/L16 will not be the same as on the C64, even if you play the same song. It's a long story but the bottom line is that the chip is clocked at different rates on the two machines (~886 vs ~985 kHz) and there is NO simple way to do differently.

That said, I don't expect different chips of the same model to sound differently, so this is a bit odd. I haven't tested many different chips, maybe 3 or 4 only, but they all worked as expected (yes, all 6581's sound slightly different from each other but they all sound good).

My references for testing the SID sound are the Lykia intro and [Solder's "Ninja" demo](https://plus4world.powweb.com/software/Synergy_SIDcard_Software).

The original SID card by Solder used the 8580 only because Solder could not come up with a working voltage step-up circuit to produce 12V (which the 6581 requires, while the 8580 takes 9V). I have an email from him where he says that, he didn't have access to the data sheet of the TL497 chip so he had to "design by attempts" and it's already a miracle he managed to get the 8580 working (he has my utmost respect for that - and well, not only for that ;)). But that is the only reason, there are no others, since the SID circuit only has minor differences on breadbins and C64Cs, and those are the same differences we account for.

The bottom line is: both SID models should be working fine in L16.

That said, what values did you use for R37 and C57?

gordonfpanam, 2025-04-17:
R37 was 100 Ohm and C57 was 470 picoFarad. I checked these against the configuration doc and these didn't have any difference. These were in the interactive bill of materials.

SukkoPera, 2025-04-17:
Can you please try R37 = 220 ohm?

gordonfpanam, 2025-05-06:
I haven’t tried the alternate passives yet, but I noticed I omitted R32 on one of the builds. I think I ran out of those and just decided I didn’t need it for the initial test.

That build sounds better, oddly, than the one with R32 populated. 

I’ll try populating R32 on the better sounding build when time permits.

SukkoPera, 2025-05-06:
R32 will just make the output slightly louder if omitted. Omission is required for the 8580 since it has a slightly lower output volume and vice-versa for the 6581. It shouldn't cause major issues in any case.

gordonfpanam, 2025-05-06:
Slightly louder is actually a good idea. The 6581 with R32 omitted still sounds somewhat quieter than the TED audio, but that might be subjective.

I'll have to record some waveforms using line input to compare the two, as recording from an iPhone camera won't do since the phone will attempt to compensate.

gordonfpanam, 2025-05-18:
I ended up replacing most of the 6581-specific caps and C52 with an aluminum one, and this seemed to improve SID audio a little bit, but replacing R37 with a 220R part improved things a lot more.

That's the SID clock, I think that makes a lot of sense that messing with that changes the audio quality.

Speaking of the SID clock, the schematic references JPs 15 and 16. I haven't set either of those, but this seems to only apply to the reset circuit and I haven't installed a reset switch on my builds yet. As such /S and /R are floating on U25A on both of my builds. Your notes say that tying /R to /RESET sounds better, did you mean audio sounding better, or being a more sound design?

SukkoPera, 2025-05-19:
Yeah, R37 is part of an RC network delaying the halved MUX signal which is used to clock the SID so that it overlaps with /CS during double-clock periods and writes can get through. It's a huge hack but any other solution is orders of magnitude more complicated. Anyway it should already be 220 in the BOM, I think I just have to fix it in the IBOM.

JP15/16 MUST be set to one value. I think we set them both to the left on our prototype and it seems to work well. If you don't do this, what might happen is that the clock signal starts up completely out of phase with /CS and everything will sound terrible because SID writes will never make it through correctly. If you don't connect the jumpers, you basically have a 50% chance of this happening at power-up and/or at every reset, so please do it. You might also test the different settings if you feel like it (there's only 2, either both to the left or both to the right, other combos don't make sense), but what I meant is "it sounds like a better idea", it will not make the sound quality better or worse, as it should either just make it work normally or totally disrupt the sound output.

gordonfpanam, 2025-06-22:
I ended up setting JPs 15 and 16 as you suggested, to the left / away from the joystick ports. I didn't notice a difference in SID audio quality, but it is working.

Replacing R37 with 250 Ohms (I didn't have any 220s handy) stabilized the 6581 behaviour on both builds.

The 6581s run rather warm on these builds compared to the same chips on a C64 longboard. I have heat sinks on them, but I'm going to try ARMSID on these when time permits. I'm hoping that solves the heat problem, not to mention it's one more modern part.

SukkoPera, 2025-06-22:
Uhm, I don't see how the SID could develop more heat, since it's running under exactly the same conditions it would be in a C64... (Clock rate apart, but that is actually _lower_...)

Anyway glad everything is working well in the end.

On a side note: there is a flaw in the joystick switch, I have just found out (fire buttons are not swapped correctly). Please disable it and consider it as if it was not there.

SukkoPera, 2025-06-22:
Fixed by 1d055ad73ee68405f88a19128d53e24c58057e10.