SukkoPera/LittleSixteen #23: 74x654 Extra Inputs from SNES Adapter on v4

gordonfpanam, 2025-05-05:
Trying to use the TexElec SNES adapter on the L16 v4 and 74x654 part seems to cause additional SNES button inputs.

Using the adapter on the original PETSCII Robots plus/4 edition seems to cause additional inputs with certain buttons. For instance, using the left shoulder button also triggers the A button input. Using a modified version from https://github.com/pdaehne/SNES-Userport-Adapter/tree/main/PETSCIIRobots-patched that accounts for third-party controllers has a slightly different effect where the left shoulder button triggers the X input instead of the A input.

Holding down A causes rapid-fire A inputs. Using the right shoulder button also triggers the left shoulder button. I haven't listed all of the odd interactions.

I just put my plus/4 back together so I can't yet verify if an original MOS 6529 part also causes this. I do know that the same controller and adapter seem to behave on an original plus/4, at least using the modified version that accounts for third-party controllers.

Diag264 reports the parallel I/O pins are working as expected.

SukkoPera, 2025-05-05:
This surpises me, since when I developed [NessieSoup](https://github.com/SukkoPera/NessieSoup), I tested it specifically on a L16 board with a CD74ACT654 without any issues. Could you please try [Joyride](https://github.com/T-Pau/Joyride)?

What value did you use for RP4? Could you please check its orientation?

gordonfpanam, 2025-05-05:
I ended up borrowing RP5 (4.7k) as I ordered too few of RP5 and RP4 for two builds. I wasn't planning on using the third joystick port immediately, and the schematics for the user port, SID and third joyport suggested that RPs 4 and 5 could be any range from 3.3k to 10k as they were being used for pullups. I figured I could omit U27 until I decided to test the third joyport, so I borrowed its RP.

I've ordered replacements for these and everything else I came up short on, but I can borrow one 8.2k from the second build to test in the meantime. I'll make sure all RPs match the bill of materials once the replacements arrive.

SukkoPera, 2025-05-05:
Yes, that makes sense and should work, as long as the orientation of the part is correct of course.

Please try Joyride.

gordonfpanam, 2025-05-05:
I'm only seeing Joyride for the C64 and C128 Native. Is there a plus/4 version out there?

SukkoPera, 2025-05-05:
Oh, you're right! That was [some wishful thinking](https://github.com/T-Pau/Joyride/issues/4).

I will provide my own test binary ASAP.

SukkoPera, 2025-05-05:
Please try the following:

[snesjtst.zip](https://github.com/user-attachments/files/20049489/snesjtst.zip)

gordonfpanam, 2025-05-06:
This tester works perfect on a stock plus/4. All of the buttons display a unique code, and button combinations come up as intended.

On both of my L16 v4 builds, I’m getting similar results, with specific buttons triggering multiple inputs, and transiting between inputs. It only happens with specific button inputs, with the rest of the buttons working as intended. 

I have video of the test and I’ll upload it when I can.

I’m going to put a 6529 back in one of the builds to make sure it isn’t something else up with the user port, but I ran out of time today. 

It turns out I was using the intended value for RP4 at 8.1k when using a x654 part, while leaving RP5 and U27 unpopulated.

gordonfpanam, 2025-05-06:
I just swapped in a 6529 to one of the builds, and the TexElec adapter is behaving better, more like the original plus/4 did. 

Now I have to find out the difference between the two.

SukkoPera, 2025-05-07:
What do you mean with "behaving better"? Is it working 100% correctly or not?

You should set the scope trigger on latch rising and then monitor clock and data, it's normal for the signals being a bit jittery. But tbh a logic analyzer is a better tool for that, most even have built-in decoders for the controller protocol.

In any case, there should be no difference in behavior between a 6529 and the '654, the circuit should make them behave equally and full compatibility is expected in both cases (and also tested on my side).

NessieSoup is also electrically equivalent to the TexElec adapter, full compatibily is expected in this case just as well. Keep in mind that there are no components on these boards, all they do is connect some userport pins straight to the controller connector pins.

SukkoPera, 2025-05-07:
Can you post a pic of your exact 654 chip?

gordonfpanam, 2025-05-07:
[image]

I’ve tried two different kinds. The ACT parts are from kits I bought from dmantione at sellmyretro. The LS parts are from a reseller in Ontario. Both behave the same.

I’ll try to post a short video showing what I’m running into.

gordonfpanam, 2025-05-07:
Here we go.

https://youtu.be/9hYZRXYKGTs

SukkoPera, 2025-05-07:
Could you please try reflowing the solder joints of U28 and those of the 654 chip?

gordonfpanam, 2025-05-07:
I'll check on U28 later today. I can reflow the joints, and I have a spare GAL16 from a TED PLA replacement kit. I don't remember what speed it was.

This sounds familiar, needing to use slower GALs on the C64 GAL PLA for certain edge cases.

gordonfpanam, 2025-05-08:
My replacement spares came in, including 4.7k RPs for RP5 and U27, so I was able to reinstall those.

Nothing directly to do with the parallel port, but I tried a 4.7k RP in RP4. The oscillating effect was more pronounced, but I could clearly see X was alternating with A on the tester this time, and the other buttons looked better still.

On a lark I soldered two 4.7k RPs to become a 2.3k RP, and put it in RP4. This seemed to fix all of the buttons.

I was able to use the 7ns GAL in U28 and still have all SNES buttons working.

I don’t yet know any other side effects of using a 2.2k RP in RP4, but I’ve used 1k pull-up resistors in other circuits with open collectors or open drains before, and didn’t see any ill effects. 

Maybe the TexElec adapter is just more sensitive to this. What other user port devices are in common use on the plus/4 that this could hurt?

SukkoPera, 2025-05-08:
Well, this is weird, there is no problem with a 2.2k pullup, I just wonder why it was not necessary on my side. The 8.2k value was calculated by Daniel Mantione for [his MOS 6529 replacement](https://www.sellmyretro.com/offer/details/43923) and we just sticked with it.

Glad you fixed the issue, let's leave this open, I guess I'll do more testing at some point...

PS: The TexElec adapter cannot be more sensitive than NessieSoup, as both of them do not contain any components, as mentioned. They just route a few userport lines to the SNES controller, you can remove the adapter altogether if you replace the SNES connector on your controller with a userport connector.

gordonfpanam, 2025-05-08:
Would you like a TexElec SNES adapter sent your way for testing?

SukkoPera, 2025-05-08:
Thanks for the offer but it's not necessary, I have just checked the schematics and they are exactly the same as NessieSoup, as I suspected.

gordonfpanam, 2025-05-18:
I couldn't find any reasonably priced 3.3k packs - right now DigiKey appears to be out of stock and I'd have to order hundreds of them. So I put together a pack of 3.3k resistors to use in RP4, and it's still very stable. This RP value works with both your SNES tester and with the original Plus/4 PETSCII Robots. The 12-bit joypad routine works, as does the 16-bit one from the patched PETSCII Robots.

So a 3.3k value for RP4 seems stable and safe enough at least for the TexElec SNES adapter. 4.7k is still a bit unstable depending on how the software reads the SNES pad. Hopefully that's a good compromise between the TexElec adapter and any other devices that would use the user port's parallel I/O.

SukkoPera, 2025-05-19:
That's some good informations, thanks. Project collaborator @kinmami is looking into this in order to shed some light, as time permits.

Anyway, 2.2k or 3.3k won't change much, I don't expect any incompatibilities. This is all about the time it takes for signals to get from the low level to the high level (it has nothing to do with the current, as some people think), since the userport has open-collector outputs. While it makes sense that some boards expect this to happen "quickly" (meaning lower pull-up resistor values), I can hardly imagine anything requiring this to be slow.

SukkoPera, 2025-05-25:
Tu further prove what @kinmami said, I did some testing applying capacitors to the clock and data lines of the controller, trying to replicate the not fully known design of the TexElec adapter. With 820 pF I can pretty much replicate the issue:

[image]
This is what happens with the 6529, everything looks normal: purple is clock generated by the computer, light blue is data being shifted out from the controller (on the clock rising edge).

[image]
This is what happens with a 74x654 with 4.7k pull-ups (what I had on hand): it looks like the controller is unable to push the line down.

I do not understand this. What I expected was that the data signal would be delayed and would not be able to get high again in time before the next bit is sampled.

Bottom line: problem is confirmed but not fully understood. If you remove the capacitors from the adapter, it should work well anyway.

gordonfpanam, 2025-05-25:
I wonder if we can invite TexElec to this conversation. There had to be some reason for him adding those caps, perhaps for the other systems this adapter was meant for.